techpaul/PS2KeyAdvanced #10: Issue with ResetKey

d-two, 2019-12-11:
I have found another small bug ;-)

After execute the "resetKey" the "PS2_led_lock" need to set zero or the LEDs will recover on the next call of setLock...

techpaul, 2019-12-11:
I will have to setup a unit and test that but I see what you mean, may have used an odd keyboard myself.

Will look at when I have some time

d-two, 2019-12-11:
I have test a little bit more and find out the "PS2_keystatus" is also not reset and the Numpad (ON/OFF) give always the same scancode back...

techpaul, 2019-12-11:
That was part of my my next testing from quick scan of code, your input helps save time

d-two, 2019-12-11:
Numpad works without problems, this was my fault 👎 ...
I use the ESP8266 and the array was not used (ARDUINO_ARCH_AVR or ARDUINO_ARCH_SAM)
https://github.com/techpaul/PS2KeyAdvanced/blob/a966b70caa59efd7bc642b50c3ee22ccbdba0345/src/PS2KeyAdvanced.cpp#L766

But with PS2_led_lock and PS2_keystatus was right...
I have add the two lines under https://github.com/techpaul/PS2KeyAdvanced/blob/a966b70caa59efd7bc642b50c3ee22ccbdba0345/src/PS2KeyAdvanced.cpp#L875 and now works all stuff after reset correct...

>// LEDs and KeyStatus Reset too...
PS2_led_lock = 0;
PS2_keystatus = 0;

Or add this line to recover LEDs and Keystatus

> // Recover LEDs
set_lock();

Both works like a charm...

PS: Sorry for my bad english ;-)

techpaul, 2019-12-12:
If you any other specific ESP details I can review them

I will look at the conditional compilation issues  and adding lines probably at weekend bit busy until then.

Thanks for the test results

techpaul, 2020-01-23:
I am bringing forward some generic updates that incorporate this due shortly

Hopefully better structure for other platform support.

techpaul, 2020-01-23:
V1.0.6 has been relased with this fix in and some other simplifications

closed 2020-01-23 by d-two