techpaul/PS2KeyAdvanced #12: Analog Input, is it possible?

meliaas, 2020-01-07:
Hey! Thanks for this project, I've been having a lot of fun tweaking this code and reusing an old keyboard that was lying around.

Do you think it's possible to tweak this to use two of the analog inputs instead of the digital inputs? Because I couldn't manage to do so, the code only works using one of the digital ports of the arduino (nano in my case).

It would be awesome so I could free up the 2 ports for other functionalities! If you could point me to any direction, I would appreciate it a lot. 

Cheers and thanks in advance.

techpaul, 2020-01-07:
If the nano supports an Analog input being reconfigured as a Digital GPIO for the Data line, however the clock line MUST be on an interrupt pin.

I do not use Nano I use many other types of boards like XMC2GO for tiny boards, 12 bit ADC, higher clock, 32 bit ARM much better RAM and Flash and actual supported in hardware I2C/SPI, roughly 1/3 cost of Nano

How are you expecting an analog pin to be used for interrupts if the board configuration does not support it, the clock rate is not fast enough to guarantee catching edges by analog or GPIO pins not interrupts and the processor is doing something else.

techpaul, 2020-01-07:
For Infineon XMC2GO see 

https://uk.farnell.com/infineon/kitxmc2goxmc1100v1tobo1/evaluation-board-cortex-m0-mcu/dp/2422880?st=xmc1100%20board 

Including VAT £6 in UK

closed 2020-01-07 by techpaul