techpaul/PS2KeyAdvanced #13: Support for STM8S mcu (SDUINO)

tinkerBOY-git, 2020-04-18:
Is STM8S mcu like the STM8S003 supported?

techpaul, 2020-04-18:
There lots and lots of boards out there.

I have not done support and full tests for that board or board series, you are welcome to try on that board but it would need your tests to tell me if it works or not.

tinkerBOY-git, 2020-04-18:
I tried to compile the **SimpleTest.ino** and I get these error:

```
WARNING: library PS2KeyAdvanced-master claims to run on avr, sam architecture(s) and may be incompatible with your current board which runs on stm8 architecture(s).
C:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc sketch\SimpleTest.ino.cpp preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\gio\Documents\Arduino\libraries\PS2KeyAdvanced-master\src -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include
Mark re12:C:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\gio\Documents\Arduino\libraries\PS2KeyAdvanced-master\src -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include sketch\SimpleTest.ino.cpp -o preproc\ctags_target_for_gcc_minus_e.cpp
cpp gefunden
C:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc sketch\SimpleTest.ino.cpp sketch\SimpleTest.ino.cpp.o re2 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\gio\Documents\Arduino\libraries\PS2KeyAdvanced-master\src -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include
Mark re2:C:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\gio\Documents\Arduino\libraries\PS2KeyAdvanced-master\src -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include sketch\SimpleTest.ino.cpp -o sketch\SimpleTest.ino.cpp.o
cpp gefunden
In file included from C:\Users\gio\Documents\Arduino\libraries\PS2KeyAdvanced-master\examples\SimpleTest\SimpleTest.ino:130:

C:/Users/gio/Documents/Arduino/libraries/PS2KeyAdvanced-master/src/PS2KeyAdvanced.h:148:2: warning: #warning Library is not supported on this board Use at own risk

C:/Users/gio/Documents/Arduino/libraries/PS2KeyAdvanced-master/src/PS2KeyAdvanced.h:349: syntax error: token -> 'PS2KeyAdvanced' ; column 20

Error compiling for board STM8S103F3 Breakout Board.
```
I just thought that there should be a support for these [STM8S boards](https://www.aliexpress.com/item/32802517941.html?spm=a2g0o.productlist.0.0.5ace6629590MPB&algo_pvid=388bda13-31ad-4b25-8169-993416cb9dab&algo_expid=388bda13-31ad-4b25-8169-993416cb9dab-0&btsid=0be3769015872075031612927e36cb&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) because they are very cheap. I'm willing to test it for you and I can also send you one or two boards if you need to.

techpaul, 2020-04-18:
Having trouble following what you are using as the install path is 

library PS2KeyAdvanced-master or
..<documents>/Arduino/Libraries/PS2KeyAdvanced-master

Which suggests it may be a fork or OLD version as errors do not match up to V1.0.7 line numbers

Please check you have the LATEST versions by

  Sketch - Include Libray -> Manage Libraries

Select Installed and see what version you have as V1.07 is the latest version

I expect the warnings but not the error you see

At the moment even if I had boards I could not look at starting full board testing for about 2 weeks
There are HUNDREDs of different boards out there and board support happens either because I did to support a board set myself or by others work being supplied. It often takles a bit of time to dela with how interrupts, flash storage, memory maps and other issues

techpaul, 2020-04-18:
What it should look like when checking library installed/version
[image]

techpaul, 2020-04-18:
For boards that size I tend to use Infineon XMC2GO boards with ARM processor, better RAM, Flash and other peripherals for around 5 to 6 GBP easily available next day

see 
https://www.digikey.co.uk/en/product-highlight/i/infineon/xmc-2go-evaluation-board
https://uk.rs-online.com/web/p/processor-microcontroller-development-kits/9106832/

or  https://github.com/Infineon/XMC-for-Arduino/wiki

tinkerBOY-git, 2020-04-18:
I'm using the latest version which is 1.0.7
[image]

I think i went looking to a lot of forums out there and have seen lots of users looking for a ps/2 library for the STM8S mcu but i never found one.

techpaul, 2020-04-18:
Then I suggest you uninstall the library and resinstall as there are things wrong, First get a clean Compile using an Arduino Uno or Mega as the board

1/ The libray is called **PS2KeyAdvanced-master** not as it should be **PS2KeyAdvanced**

2/ Your first error says 
C:/Users/gio/Documents/Arduino/libraries/PS2KeyAdvanced-master/src/PS2KeyAdvanced.h:148:2: warning: #warning Library is not supported on this board Use at own risk
This is on the **WRONG** line for V1.07 as the lines are
~~~
148  #if defined( ARDUINO_ARCH_SAM )
149  #define PS2_SUPPORTED           1
     ....
158  // Invalid architecture
159  #if !( defined( PS2_SUPPORTED ) )
160  #warning Library is NOT supported on this board Use at your OWN risk
161  #endif
~~~

So error should have been reported on 160 not 148

3/ Your second error says
C:/Users/gio/Documents/Arduino/libraries/PS2KeyAdvanced-master/src/PS2KeyAdvanced.h:349: syntax error: token -> 'PS2KeyAdvanced' ; column 20

Line 349 of V1.0.7 is 
#define PS2_KEY_LANG1       0X96

When I compile the SAME sketch for Arduino Mega 256 The ouputs I get (some lines fopr core removed)

~~~
Using board 'mega' from platform in folder: C:\....\Arduino15\packages\arduino\hardware\avr\1.8.2
Using core 'arduino' from platform in folder: C:\....\Arduino15\packages\arduino\hardware\avr\1.8.2
Detecting libraries used...
"C:\\....\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\....\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\cores\\arduino" "-IC:\\....\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\variants\\mega" "C:\\....\\Temp\\arduino_build_897041\\sketch\\SimpleTest.ino.cpp" -o nul
"C:\\....\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\....\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\cores\\arduino" "-IC:\\....\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\variants\\mega" "-IC:\\....\\Documents\\Arduino\\libraries\\PS2KeyAdvanced\\src" "C:\\....\\Temp\\arduino_build_897041\\sketch\\SimpleTest.ino.cpp" -o nul
"C:\\....\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\....\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\cores\\arduino" "-IC:\\....\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\variants\\mega" "-IC:\\....\\Documents\\Arduino\\libraries\\PS2KeyAdvanced\\src" "C:\\....\\Arduino\\libraries\\PS2KeyAdvanced\\src\\PS2KeyAdvanced.cpp" -o nul
Generating function prototypes...
"C:\\....\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\....\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\cores\\arduino" "-IC:\\....\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\variants\\mega" "-IC:\\....\\Documents\\Arduino\\libraries\\PS2KeyAdvanced\\src" "C:\\....\\Temp\\arduino_build_897041\\sketch\\SimpleTest.ino.cpp" -o "C:\\....\\Temp\\arduino_build_897041\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\....\\Temp\\arduino_build_897041\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\....\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\....\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\cores\\arduino" "-IC:\\....\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\variants\\mega" "-IC:\\....\\Documents\\Arduino\\libraries\\PS2KeyAdvanced\\src" "C:\\....\\Temp\\arduino_build_897041\\sketch\\SimpleTest.ino.cpp" -o "C:\\....\\Temp\\arduino_build_897041\\sketch\\SimpleTest.ino.cpp.o"
Compiling libraries...
Compiling library "PS2KeyAdvanced"
"C:\\....\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\....\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.2\\cores\\arduino" "-IC:\\....\\Arduino15\\packages\\arduino\\hardware\\ [... cut ...]
Using library PS2KeyAdvanced at version 1.0.7 in folder: C:\....\Documents\Arduino\libraries\PS2KeyAdvanced 
"C:\\....\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-size" -A "C:\\....\\Temp\\arduino_build_897041/SimpleTest.ino.elf"
Sketch uses 6046 bytes (2%) of program storage space. Maximum is 253952 bytes.
Global variables use 323 bytes (3%) of dynamic memory, leaving 7869 bytes for local variables. Maximum is 8192 bytes.
~~~

You appear to have installation errors fro the library or other issues to get over first

tinkerBOY-git, 2020-04-19:
So i tried installing fresh Arduino, PS2KeyAdvanced, and sduino on a different pc but i think the errors are the same.

```
WARNING: library PS2KeyAdvanced-master claims to run on avr, sam, samd1 architecture(s) and may be incompatible with your current board which runs on stm8 architecture(s).
C:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc sketch\SimpleTest.ino.cpp preproc\ctags_target_for_gcc_minus_e.cpp re12 -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\gio\Documents\Arduino\libraries\PS2KeyAdvanced-master\src -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include
Mark re12:C:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ -E -MC -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\gio\Documents\Arduino\libraries\PS2KeyAdvanced-master\src -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include sketch\SimpleTest.ino.cpp -o preproc\ctags_target_for_gcc_minus_e.cpp
cpp gefunden
C:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc sketch\SimpleTest.ino.cpp sketch\SimpleTest.ino.cpp.o re2 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\gio\Documents\Arduino\libraries\PS2KeyAdvanced-master\src -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include
Mark re2:C:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/bin/sdcc -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mstm8 -DSTM8S103 -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_STM8S_BLUE -DARDUINO_ARCH_STM8 -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\cores\sduino -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0\variants\standard -IC:\Users\gio\Documents\Arduino\libraries\PS2KeyAdvanced-master\src -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\hardware\stm8\0.5.0/STM8S_StdPeriph_Driver/inc -IC:\Users\gio\AppData\Local\Arduino15\packages\sduino\tools\sdcc\build.11242/include sketch\SimpleTest.ino.cpp -o sketch\SimpleTest.ino.cpp.o
cpp gefunden
In file included from C:\Users\gio\Documents\Arduino\libraries\PS2KeyAdvanced-master\examples\SimpleTest\SimpleTest.ino:130:

C:/Users/gio/Documents/Arduino/libraries/PS2KeyAdvanced-master/src/PS2KeyAdvanced.h:160:2: warning: #warning Library is NOT supported on this board Use at your OWN risk

C:/Users/gio/Documents/Arduino/libraries/PS2KeyAdvanced-master/src/PS2KeyAdvanced.h:361: syntax error: token -> 'PS2KeyAdvanced' ; column 20

exit status 1
Error compiling for board STM8S103F3 Breakout Board.
```

techpaul, 2020-04-19:
First of all the errors are DIFFERENT but similar the line numbers in the error reports show DIFFERENT versions of files being used. One warning error is expected when using an unsupported board the other should NOT happen if ALL the compiler software is correctly installed and properly does C++ as the second line is a standard C++ class declaration. If your compiler is removing class keyword or similar it will NOT work with most Arduino software.

The library is installable from Arduino Library manager via the IDE and many hundreds of people have managed that, to work with supported boards. On an unsupported board the warnings should be -

From the IDE
~~~
WARNING: library PS2KeyAdvanced claims to run on (avr, sam, samd1) architecture(s) and may be incompatible with your current board which runs on (arm) architecture(s).
~~~
Note the name is PS2KeyAdvanced NOT anything else

From compiling the library header
~~~
C:\...\Documents\Arduino\libraries\PS2KeyAdvanced\src/PS2KeyAdvanced.h:160:2: warning: #warning Library is NOT supported on this board Use at your OWN risk [-Wcpp]

 #warning Library is NOT supported on this board Use at your OWN risk

  ^
~~~
Your first installation said a different line number
~~~
C:/.../Documents/Arduino/libraries/PS2KeyAdvanced-master/src/PS2KeyAdvanced.h:148:2: warning: #warning Library is not supported on this board Use at own risk
~~~
That is PS2KeyAdvanced.h:148:2: 

Which means 
~~~
File    PS2KeyAdvanced.h
Line    148
Column  2 
~~~
You have the same errors on different line numbers showing incorrect library installation amongst other things, between systems.

Steps to do first

1.  Remove existing library
2.  Use Arduino library manager to load the library at latest version, use the filter box with "PS2Kety*" to find it quicker
3.  Set board to ARDUINO UNO
4.  Do a Verify/compile  by  Sketch -> Verify/Compile
    or Top left button on toolbar
[image]

This will COMPILE ONLY the sketch and if installed correctly give NO errors, This method does NOT need any board to test The messages should also show that the library PS2KeyAdvanced and no other name is used.

Until you can do that, there is no point chasing anything else down

closed 2020-11-01 by techpaul