techpaul/PS2KeyAdvanced #14: Toggle typematic on/off entirely?

inchoat3, 2020-10-11:
Hi there,
I'm new to using GitHub; hopefully I'm in the right spot.

I've been attempting to use your library as part of a MIDI controller, but the typematic repeat is problematic when it comes to controlling an instrument, so I set about trying to send a 0xF8 command to set it to make/break only. I'm just not sure how to get the keyboard to receive the command, as everywhere I've tried to edge the command in causes it to glitch out or does nothing.

I've added this at line 106 of PS2KeyCode.h:
#define PS2_KC_MKBRK    0xF8

this at line 294 of PS2KeyAdvanced.h:
void mkBrk( void );

and this at 967 of PS2KeyAdvanced.cpp:
void PS2KeyAdvanced::mkBrk( )
{
  send_byte( PS2_KC_MKBRK );            // send command
  if ( ( send_byte( PS2_KEY_IGNORE ) ) ) // wait data PS2_KC_BAT or PS2_KC_ERROR
    send_next( );                        // if idle start transmission
}

When I try to pull that class either in setup or in the loop it still doesn't work. I also tried to inject a command into the resetKey command with no success.

If you take feature requests, I'd sure love to see it in there.  You've done great work here; it's great for breathing new life in to old equipment.

Thanks for any info you might have.

techpaul, 2020-10-11:
In the library folders there is a document PS2 Keyboard.pdf on page 4 there 
is anote at bottom of page

	 The next six commands can be issued when the keyboard is in any
	 mode, but it only effects the behavior of the keyboard when in
	 "mode 3" (ie, set to scan code set 3.)

The library only works with Scan Code Set 2, the function you want if
supported by the keyboard ONLY work in scan code set 3. It is rare for
keyboards these days to work in scan code set 1 or 3.

Other alternative solutions

Set Typematic rate to

	Delay 1 sec
	repeat 2 cps

	Values sent  0xF3, 0x7F

Realistically higher code should keep track of make/break status and
handle how to denote changes to higher functions, so repeated SLOW rate of
repeats are ignored, then clear current known key from list of keys in make
state.

This is not a realistic feature to add at library level.

inchoat3, 2020-10-11:
That makes a lot more sense, and clears up why another implementation I saw was running set 3. I appreciate your response and will be adjusting my approach.

closed 2020-10-11 by inchoat3