techpaul/PS2KeyAdvanced #16: Acknowledging additional startup PS/2->Host messages

SteveGuidi, 2021-02-10:
I suspect this is not a wide-spread startup handshake for all PS/2 keyboards, but it appears to be happening to at least three keyboards in the wild (mine, a YouTube user's, and a fellow from an Arduino forum).  I'm curious it if is feasible to add support for this in the `begin()` method.

The library reports that a constant sequence of 0xAA messages are sent by the keyboard, which appears to happen during or after the self-test completes (LED lights on keyboard go off).  According to [this post from 2011](https://forum.arduino.cc/index.php?topic=57610.msg414796#msg414796), the sequence has an incorrect parity bit set (presumably to distinguish from the standard BAT 0xAA message), and needs to be acknowledged by the 0xFE message, also with an incorrect parity bit set.

This is my first Arduino project, and I'm slowly understanding the platform.  I could probably submit a pull request for this once I figure out how to implement the acknowledgement; you might not have a keyboard to test this with.  For what its worth, my keyboard has a dual USB-HID and PS/2 controller.

techpaul, 2021-02-10:
First of all there is a document ijn the library folder called PS2 Keyboard.pdf (also viewable from here under the code section).

NO on the many keyboards myself and ohers who have used this library have found there is NO multiple or constant 0xAA, there is ONE. with CORRECT parity.

To get multiple 0xAA whether oarity right or wrong is a DIFFERENT fault, normally bad power connections or power dipping because keyboard draws too much power from +5V. Causing keyboard to reset multiple times. Either that or you have a very old XT keyboard with different protocol, or it is using a strange NOT PS2 mode or a very strange Scan Code set.

0xAA is ONLY used for Power Up and Self Test PASSED with CORRECT parity.

0xFE (from either end) is resend last byte and not acknowledge for PS2 keyboards Usually resend due to parity ERRORS.

Lights/LEDs going On then Off is NOT part of PS2 keyboard initialisation, it is a Windows/BIOS/UEFI action to SEND commands to do this, see document mentioned above, page 17 Initialisation from an i8042 controller in  a PC.

This is the FIRST time someone claimed multiple 0xAA at power up.

I suspect strange or faulty keyboard or faulty power (wiring or not enough current available).

SteveGuidi, 2021-02-10:
Thank you for the feedback.  I'll continue to analyze this set of messages on my keyboard and try to pinpoint its origin.

closed 2021-02-10 by SteveGuidi

skrinsky, 2021-02-18:
Hey @SteveGuidi I'm having the same issue- did you find a solution?

skrinsky, 2021-02-18:
Hey @SteveGuidi I am having the same issue, did you find a solution?

SteveGuidi, 2021-02-18:
@skrinsky Sorry, I have not had a chance to resume working my project yet.  I would like to try the suggestion of powering the keyboard from a regulated 5V supply externally next.  I'm using the regulated supply on the Arduino Nano right now, and although it should be capable of providing the required 100mA, testing with an external supply will help eliminate variables.

For what its worth, my keyboard is very likely a dual PS/2 and USB keyboard.  I have replaced the PS/2 connector with a USB-A connector (wired as if the PS/2 connector were going through a passive adapter) and my computer detects and enables it without any problems.