techpaul/PS2KeyAdvanced #17: CODE AA error

skrinsky, 2021-02-18:
Hi, I am getting a repeating of the "VALUE AA - Status Bits 0 Code AA" error message, which is supposed to correspond to an error where the keyboard is attached after the program began running, however this keyboard was attached while I uploaded the code. Help is much appreciated thanks!

skrinsky, 2021-02-18:
just read someone else was having a similar problem, wanted to specify this is the keyboard I am using, just got two of them brand new and have the same problem with both
https://www.amazon.com/Perixx-PERIBOARD-409P-Mini-Keyboard-12-40x5-79x0-79/dp/B00JV08TIA/ref=sr_1_3?dchild=1&keywords=keyboard+mini+ps%2F2&qid=1613675064&sr=8-3

techpaul, 2021-02-18:
Code 0xAA is the power up test PASS of the BAT Basic Assurance Test.

This message is ONLY sent when

* Keyboard powered up
* Keyboard received a RESET Command sent from the host of 0xFF

This is powered up and ready status from the keyboard.

If you are getting REPEATED 0xAA I would srongly suggest you check the power as it is dipping and reseting the keyboard.

Power your Arduino from an external power supply as there is probably not enough power from the USB power to drive the arduino and the keyboard and any other connected circuitry.

techpaul, 2021-02-18:
USB keyboards with PS2 adapters draw TOO MUCH POWER from and Arduino to operate correctly.

skrinsky, 2021-02-19:
Gotcha, these keyboards claim to be PS2 but in reviews I found someone saying "IF YOU ARE TRYING TO BUY A NATIVE PS/2 KEYBOARD, STAY AWAY FROM THIS. THIS IS A USB KEYBOARD WITH A PS/2 CONNECTOR AND AS A RESULT DOES NOT WORK WITH A TRUE NATIVE PS/2 INPUT. " bummer! external power supply worked. I have to run this project with batteries anyways eventually so should be all good because I assume a 9v should be enough... thanks for the help.

drencorxeen, 2021-02-19:
@skrinsky ,
I can confirm that those keyboards linked at the Amazon URL in fact do not work correctly.  I had ordered a couple of them and had to return them as they would not work on a old Gateway 2000 486DX2/66 system nor would it work on a true IBM PS/2 system.

I did some testing using a Arduino unit and a 6PIN mini DIN to test the keyboards using this library and in fact the keyboard always reports an 0xAA and the entire project was powered via a 5V 3A supply.

It is sad as the feel of the keyboards are nice, but what ever µCU they used in the keyboards must not have had a good firmware loaded in them to correctly support the PS/2 protocol.

SteveGuidi, 2021-02-22:
I'm going to put my notes here as I originally opened #16 a few weeks ago.

Coincidentally, I am also using the same Periboard keyboard!  Thank you @skrinsky and @drencorxeen for confirming that this is indeed a USB keyboard rewired for PS/2 behavior.  I suspect this when I opened up the keyboard and saw several unpopulated USB hub PCB footprints (indicating a reused PCB for different models).  I replaced the PS/2 connector with a male USB-A and Windows detects the device as an HID keyboard.

In #16, I noted that I was starting to test the keyboard using an external power supply.  When I did this, the repeated 0xAA issue went away, although I still can't read inputs from the keyboard (presumably because of the USB design issue above).  I can also confirm that repeated 0xAA issue arises only when the device is powered from the Arduino's regulator, which confirms an earlier note from @techpaul about too much current draw causing the keyboard to constantly reset.

So, the solution is to find a proper PS/2 keyboard!

closed 2021-02-28 by skrinsky