techpaul/PS2KeyAdvanced #18: Hold CAPS

EPecherkin, 2021-05-02:
Great library! I found it particularly useful in my project.
I noticed that CAPS cannot be held. The library sends 3 events in a row once it's pressed, and 0 when it is released.
I would like to find a way to catch it when it is released. (or I need to remap it to LCTRL but with the ability to catch the "release" event too).
How can I achieve this?

techpaul, 2021-05-03:
I am having trouble understanding what you mean as you are using conflicting terms.

Repetitions of make codes (when key is pressed) is standard because this is the Typematic repeat performed by the keyboard, you can set the library to ignore repeats on SOME keys like CTRL, SHIFT, CAPS LOCK, ALT, SCROLL LOCK, NUM LOCK, if that is what YOU wish

The **bottom 8** bits is always equivalent to ASCII upper case for A to Z, it is the **upper 8** bits that determine what the status of make/break and other states like simultaneous keys Shift, Ctrl, Alt or GUI are pressed at same. Remembering CAPS is done by the **CAPS LOCK** key not any other key. Once pressed the CAPS LOCK LED is lit, until next time Caps Lock is pressed.

When you say "LCTRL" I assume you mean Left Control (CTRL) key which is a **DIFFERENT** function.

See folder in your installed library called "extra" for the documentation

There is a bit in the returned code that says if this is a Make or Break code, the following constants tell you different things

| Constant | | Description |
|---|-|---|
|PS2_BREAK| |This is a key break event (key release) |
|PS2_SHIFT| |One of the SHIFT keys is STILL pressed at same time |
|PS2_CTRL| |One of the CTRL keys is STILL pressed at same time |
|PS2_ALT| |The (left) ALT key is STILL pressed at same time |
|PS2_ALT_GR| |The (right) ALT or ALT-GR key is STILL pressed at same time |
|PS2_GUI| |One of the GUI keys is STILL pressed at same time |
|PS2_CAPS| | CAPS LOCK mode is on |
|PS2_FUNCTION| | Keycode is a NON-displayable code function code |

To capture release events you **MUST** remember that the returned value form the library is an **unsigned 16 bit int (uint16_t)**, to check for release code you need to **AND** the value with PS2_BREAK to see if this is a release event.

Key code values for BOTTOM 8 bits on functions like Shift are
~~~
    PS2_KEY_NUM
    PS2_KEY_SCROLL
    PS2_KEY_CAPS
    PS2_KEY_PRTSCR
    PS2_KEY_PAUSE
    PS2_KEY_L_SHIFT
    PS2_KEY_R_SHIFT
    PS2_KEY_L_CTRL
    PS2_KEY_R_CTRL
    PS2_KEY_L_ALT
    PS2_KEY_R_ALT
    PS2_KEY_L_GUI   (Sometimes called windows key)
    PS2_KEY_R_GUI   (Sometimes called windows key)
    PS2_KEY_MENU
~~~

Your description of problem suggests either misunderstanding of keyboard operations or not handling the values returned from the library correctly.

EPecherkin, 2021-05-06:
Yes, all that is right. Therefore there is a difference from the original behavior (if I connect keyboard directly to the computer).
Let's talk about computer games. You can map your CAPSLOCK key to `Sit` for example. And why you hold it - you sit. You release it - you stand up. With the current implementation of the library - I can't get the same behavior because there is simply no event when the CAPSLOCK key is released. Only when pressed.
I would like to find a way to catch an event when CAPSLOCK is released OR change the behavior of the CAPSLOCK key to act as LEFT CTRL (with press/release events of course)

techpaul, 2021-05-07:
Can I suggest you should probably have either used the simpler RAW library https://github.com/techpaul/PS2KeyRaw or
you are welcome to fork a branch to change operation.

Vast majority os users are looking for ways to remove confusion on decoding every event and get filtered data in easier to understand data form to their application or add internationalisation.

closed 2021-06-22 by techpaul