techpaul/PS2KeyAdvanced #19: Host-to-Device: Initial Clock low too short

Schubidubida, 2021-05-14:
Function send_now():504 waits for 60 microseconds:
delayMicroseconds( 60 );

This freezes my trusty old Multitech KB084 as soon as an LED is set (CAPS, SCROLL, NUM). 
In the PDF from the extras folder, a different time is specified:

"Inhibit communication by pulling Clock low for at least 100 microseconds."

By changing the above line to 

delayMicroseconds( 120 );

everything is working perfectly.

techpaul, 2021-05-15:
Please be aware that your old keyboard may be expecting a 150 to 200 microsecond pulse.

Measurements with a scope of the actual signal with shorter delay should give a 120 microsecond pulse, due to time to execute code following that line.

Electrical timing measurements were taken using boards such as UNO and DUE, before initial release

Schubidubida, 2021-05-15:
On 15/05/2021 11:38, Paul Carpenter wrote:
> Please be aware that your old keyboard may be expecting a 150 to 200 microsecond pulse.

That might be the case, it is REALLY old.
> Measurements with a scope of the actual signal with shorter delay should give a 120 microsecond pulse, due to time to 
> execute code following that line.

I understand.

> Electrical timing measurements were taken using boards such as UNO and DUE, before initial release

I'm pretty sure that Your sketch is working for 99.9% of all the old PS/2 Keyboards out there :-) Mine is a somehow 
exotic beast, it won't work with any commercial ps/2 - USB adapter.

Best Regards,

Christof

-- 
Mit freundlichen Grüßen,

C. Reetz

------------------

Christof J. Reetz

CRUSE Spezialmaschinen GmbH
Gimmersdorfer Straße 76
53343 Wachtberg
Germany

Fon	+49 228 933975-0
Fax	+49 228 933975-29
Web	https://www.crusescanner.com

The Text used in this eMail is made from 100% recycled electrons and magnetic particles.
Keep it away from sunlight, pets, and small children.

techpaul, 2021-05-15:
PS2 to USB adapters  issues depends on the USB keybpard and the adapter as to how much current they draw. Most USB keyboards draw more power than can be supplied from Arduino boards on USB power. This is a POWER issue with USB keyboards relying on 90 to 200mA being available just for the keyboard. Some PS2 to USB adapters are active and draw more power as well.

closed 2021-06-22 by techpaul