techpaul/PS2KeyAdvanced #25: Send own bytes to keyboard

steveexe, 2021-09-11:
Hello.
 I have on my hands Creative ProdikeysDM. It's PS/2 keyboard, but i need using this on USB (because drivers available only on WinXP or Win7x32). Thankfuly, i found page (https://linuxmusicians.com/viewtopic.php?t=15947), where one guy decided, how to work with that. He even create project on Git, where he using Arduino Uno R3. I think using Arduino Micro instead to place inside case.
 I try write my onw code to work with PS/2 protocol, but fail (i can only receving data, but cannot send, idk why). Also i found this library and i notice function "send_byte", but not available from class.
 It is possible make this function public?

techpaul, 2021-09-11:
Have you got basice PS2 Keyboard functionality working with ability to send the functions the library supports (LED lighting typematic rate adjust etc).

Sounds to me from looking at the data you need a complex set of libraries that BLENDS MIDI and PS2 interface, so called programmeable keys are more likely programmeable response oin the HOST from standard key presses like F! to F24.

I think you are looking at this the wrong way round, to get the PS2 Keyboard working does not require you to know how to send_byte, as there are other aspects of the protocol you have to handle beyond that. I suggest finding out how you interface the the other aspects like pitch bend and notes using PS2 interface first.

You can fork this repository to your own and do what you like with it there. What you are requiring is beyond scope of this project.

cyberluke, 2021-09-11:
Look up some usb hid arduino/teensy programming. Try to search for ps/2 usb
router project or ps/2 usb proxy. I use Teensy 3.2 and recieve usb gaming
keyboard data over usn host, then send it to ps/2 using this library. PS/2
is standard typing keyboard interface. From usb hid standpoint, I sent
something called hid report, which is some sort of byte packet. Some usb
hosts let you send this easily using Keyboard.send_now() command and
sending up to 8 keys at once. If you need something special, you need press
your key and print it in serial console, some int code, etc. Then you can
send it. So you need usb host, which you can use as usb analyzer as well.
Enjoy few weekends!

techpaul, 2021-10-02:
No further answers from original poster so closing

closed 2021-10-02 by techpaul