techpaul/PS2KeyAdvanced #27: 2 keyboards in 1 board

RiNoize, 2022-02-02:
Hello Paul, I would like to know if it can be connected and how, 2 (two) Ps2 keyboards to the same board, I was trying but I can't make them both work at the same time.  Thanks

techpaul, 2022-02-02:
Without knowing what board you are using and EACH keyboard has its OWN interrupt pin, it is difficult to see what your problem is.

RiNoize, 2022-02-02:
I tried with an Arduino mega pro (1 keyboard on pins 20 and 21 and the other keyboard on pins 2 and 3) and with an arduino pro micro (on pins 16 and 2 and the other keyboard on pins 15 and 3).
With 1 keyboard at a time, any of the two keyboards works perfectly, the issue is with both connected and declared at the same time.
I declared them this way in the pro micro:

#define DATAPIN 16
#define IRQPIN 2

#define DATAPIN2 15
#define IRQPIN2 3
uint16_t c;
uint8_t breaks = 0;
uint8_t repeats = 0;

PS2KeyAdvanced keyboard;
PS2KeyAdvanced keyboard2;

CoreyOsgood, 2023-11-25:
I'm seeing a similar issue with an ESP32 and 2 keyboards. Comment out the code and individually either works fine. If both are set up, only keyboard #2 works. Most of the time the keypresses end up tripping keyboard1.available() and are read as if they came from keyboard #1, although sometimes it does come through on keyboard2. Keypresses on Keyboard #1 don't do anything.

[DualKeyboardESP32Test.txt](https://github.com/techpaul/PS2KeyAdvanced/files/13465415/DualKeyboardESP32Test.txt)